UpstreamData/pyasic #350: Support for Avalon Q Home

mark0st83, 2025-06-19:
Need Support for the new Canaan AvalonQ Home.

See the Miner HomeAssistant Integration 

https://github.com/Schnitzel/hass-miner/issues/545#issue-3134306221

R1cco, 2025-06-30:
Yes pls!

c7ph3r10, 2025-07-04:
> I was just jumping on to say the same thing. Received one of these today and would like to be able to automate it around solar production etc.

I found a workaround for myself. I documented it in the hass-miner Issues chapter which is linked above. May be you find it helpful as well. The workaround is not home assistant specific.

c7ph3r10, 2025-07-22:
> Ok, trying to get around to doing this now, but it seems like they change commands every few versions of miner, so I don't know if what I already have will be consistent. Most of the basic data gathering seems standard, but I'm guessing the `softon`/`softoff` commands will need to be tested in order to implement pause/resume.
> 
> Would anyone like to try replacing the timestamp parameter in those commands with `0`? I want to see if it will accept a timestamp in the past or not, if not we may need to rely on the result of another command from the miner to have an accurate timestamp.
> 
> If anyone has one of these and wants help putting together a PR for them, let me know and I can help some more, but this is likely going to be one that I can't do easily without access to one.

Hey!
In case you need some help in testing something may be I can support in this case. I have three Avalon Qs running with a semi professional load management based on home assistant (see link in the initial post).

Canaan Avalon Q behaves as following regarding softon/softoff:

Szenario 1:
Avalon Q has no power although power supply switch is in "on" state. Activation of the power for the power supply makes the device automatically switch to the latest mode (Eco/Standard/Super) it was in when it was turned off. Even if it was send to standby bevore the power supply was disconnected from power it will start with the latest workmode it ran before it was disconnected from power. This szenario has nothing to to with the commands mentioned within my hass-miner post. It´s just to get the full picture.

Szenario 2:
Avalon Q is in workmode Eco/Standard/Super. Sending softoff with timestamp '0'
`echo 'ascset|0,softoff,1:0' | nc 10.0.0.201 4028`
leads to this response:
`STATUS=E,When=1753122013,Code=120,Msg=ASC 0 set failed: failed softoff:time interval is too short,Description=cgminer 4.11.1|#`
So this does not work.
What works is:
`/bin/bash -c "echo 'ascset|0,softoff,1:$(( $(date +%s) + 5 ))' | nc 10.0.0.201 4028"`
So use current unix timestamp and add some secods (e.g. 5). Response looks like this:
`STATUS=I,When=1753122505,Code=118,Msg=ASC 0 set info: success softoff:1753122510,Description=cgminer 4.11.1|#`

Szenario 3:
Avalon Q was sent to standby via softoff. Power supply is still connected to power.
Sending softon with timestemp '0':
`echo 'ascset|0,softon,1:0' | nc 10.0.0.201 4028`
leads to this response:
`STATUS=I,When=1753158262,Code=118,Msg=ASC 0 set info: success softon:0,Description=cgminer 4.11.1|#`
so this gives a success response but for whatever reason the miner does NOT start!
What also does work is:
`/bin/bash -c "echo 'ascset|0,softon,1:$(( $(date +%s) + 5 ))' | nc 10.0.0.201 4028"`
So using unix timestamp solves the problem also with softon. Response is:
`STATUS=I,When=1753158467,Code=118,Msg=ASC 0 set info: success softon:1753158472,Description=cgminer 4.11.1|# `

Szenario 4:
Avalon Q is in standby and it shall be switched to a workmode (Eco/Standard/Super).
Sending command:
`echo 'ascset|0,workmode,set,1' | nc 10.0.0.201 4028`
Response:
`STATUS=E,When=1753158766,Code=120,Msg=ASC 0 set failed: idle or current mode 0 is caling. Don't permit switch mode,Description=cgminer 4.11.1|#  `
Conclusion: The miner always needs a softon ahead of using any workmode. Between softon and setting the workmode a delay should be set. In my configuration I use 45 seconds to be on the safe side.

I hope this helps to get softon/soft off managed. If you need further support let mit know. I have three of those running.

Here are some further commands and their responses. May be this helps to get some needed states:
Command:
`echo "summary|0" | nc 10.0.0.201 4028`
Response
`STATUS=S,When=1753159103,Code=11,Msg=Summary,Description=cgminer 4.11.1|SUMMARY,Elapsed=995,MHS av=6187974.81,MHS 5s=0.00,MHS 1m=17746.51,MHS 5m=2693620.78,MHS 15m=3347809.61,Found Blocks=0,Getworks=22,Accepted=11,Rejected=0,Hardware Errors=0,Utility=0.66,Discarded=0,Stale=0,Get Failures=0,Local Work=706,Remote Failures=0,Network Blocks=4,Total MH=6157265100.0000,Work Utility=86445.01,Difficulty Accepted=1286432.00000000,Difficulty Rejected=0.00000000,Difficulty Stale=0.00000000,Best Share=1719834,Device Hardware%=0.0000,Device Rejected%=0.0000,Pool Rejected%=0.0000,Pool Stale%=0.0000,Last getwork=0|# `

Command:
`echo "stats|0" | nc 10.0.0.201 4028`
Response:
`STATUS=S,When=1753159231,Code=70,Msg=CGMiner stats,Description=cgminer 4.11.1|STATS=0,ID=AVALON0,Elapsed=1128,Calls=0,Wait=0.000000,Max=0.000000,Min=99999999.000000,MM ID0:Summary=,HBinfo=|STATS=1,ID=POOL0,Elapsed=1128,Calls=0,Wait=0.000000,Max=0.000000,Min=99999999.000000,Pool Calls=0,Pool Attempts=0,Pool Wait=0.000000,Pool Max=0.000000,Pool Min=99999999.000000,Pool Av=0.000000,Work Had Roll Time=false,Work Can Roll=false,Work Had Expire=false,Work Roll Time=0,Work Diff=65536.00000000,Min Diff=65536.00000000,Max Diff=262144.00000000,Min Diff Count=3,Max Diff Count=1,Times Sent=30,Bytes Sent=4686,Times Recv=57,Bytes Recv=29573,Net Bytes Sent=4686,Net Bytes Recv=29573|#`

Command:
`echo "litestats|0" | nc 10.0.0.201 4028`
Response:
`STATUS=S,When=1753159324,Code=70,Msg=CGMiner stats,Description=cgminer 4.11.1|MM ID0:Summary='STATS':{Ver[Q-25052801_14a19a2] LVer[25052801_14a19a2] BVer[25052801_14a19a2] HashMcu0Ver[Q_hb_v1.1] FanMcuVer[Q_fb_v1.2] CPU[K230] FW[Release] DNA[0201000018f85eaf] STATE[1] MEMFREE[67956] NETFAIL[0 0 0 0 0 0 0 0] SSID[] RSSI[0] NetDevType[0] SYSTEMSTATU[Work: In Work, Hash Board: 1 [... cut ...]
`STATUS=S,When=1753159658,Code=72,Msg=Check command,Description=cgminer 4.11.1|CHECK,Exists=N,Access=N|# `

Command:
`echo "version|0" | nc 10.0.0.201 4028`
Response:
`STATUS=S,When=1753159712,Code=22,Msg=CGMiner versions,Description=cgminer 4.11.1|VERSION,CGMiner=4.11.1,API=3.7,PROD=Avalon Q,MODEL=Q,HWTYPE=Q_MM1v1_X1,SWTYPE=MM319,LVERSION=25052801_14a19a2,BVERSION=25052801_14a19a2,CGVERSION=25052801_14a19a2,HBMCUVERSION=Q_hb_v1.1,FANMCUVERSION=Q_fb_v1.2,DNA=0201000018f85eaf,MAC=00e04cce67a0|#`

That is all I got so far. As said. If you need further support, let me know.

b-rowan, 2025-07-22:
For context, the reason I prefer JSON is because pyasic uses that style internally so it is much easier to parse, plus I can run tests against it.  There is also the case of the `litestats` command, which used to be just `stats`, which returns jumbled garbage as part of the response, making it much much harder to parse if it isn't in JSON.

c7ph3r10, 2025-07-22:
> here is also the case of the `litestats` command, which used to be just `stats`

The following command does its work:
`echo '{"command": "litestats", "param": "0"}' | nc 10.0.0.201 4028`

JSON response:
{"STATUS":[{"STATUS":"S","When":1753220531,"Code":70,"Msg":"CGMiner stats","Description":"cgminer 4.11.1"}],"STATS":[{"MM ID0:Summary":"'STATS':{Ver[Q-25052801_14a19a2] LVer[25052801_14a19a2] BVer[25052801_14a19a2] HashMcu0Ver[Q_hb_v1.1] FanMcuVer[Q_fb_v1.2] CPU[K230] FW[Release] DNA[0201000018f85eaf] STATE[1] MEMFREE[67952] NETFAIL[0 0 0 0 0 0 0 0] SSID[] RSSI[0] NetDevType[0] SYSTEMSTATU[Work: In Work, Hash Board: 1] Elapsed[1412] BOOTBY[0x01.00000000] LW[737509] MH[0] DHW[0] HW[0] DH[2.337%] ITemp[32] HBITemp[60] HBOTemp[63] TMax[72] TAvg[65] TarT[65] Fan1[1401] Fan2[1373] Fan3[1386] Fan4[1415] FanR[44%] SoftOffTime[10] SoftOnTime[0] Filter[16882] FanErr[0] SoloAllowed[1] PS[0 1217 2250 35 808 2251 834] PCOMM_E[0] GHSspd[54146.67] DHspd[2.337%] GHSmm[55445.83] GHSavg[51528.37] WU[719842.95] Freq[284.98] MGHS[51528.37] TA[160] Core[A3197S] BIN[36] PING[17] SoftOFF[0] ECHU[0] ECMM[0] PLL0[7488 5732 5388 5712] SF0[258 276 297 318] CRC[0] COMCRC[0] ATA0[800-65-2264-258-20] LcdOnoff[1] Activation[0] WORKMODE[0] WORKLEVEL[0] MPO[800] CALIALL[7] ADJ[1] Nonce Mask[25]}"}],"id":1}#

b-rowan, 2025-07-22:
`param` is `,` separated values, device id is likely the first param...

c7ph3r10, 2025-07-22:
guess you prefer this one:

`echo '{"command": "stats", "param": "0"}' | nc 10.0.0.201 4028`

response:
`{"STATUS":[{"STATUS":"S","When":1753220674,"Code":70,"Msg":"CGMiner stats","Description":"cgminer 4.11.1"}],"STATS":[{"STATS":0,"ID":"AVALON0","Elapsed":1528,"Calls":0,"Wait":0.000000,"Max":0.000000,"Min":99999999.000000,"MM ID0:Summary":"","HBinfo":""},{"STATS":1,"ID":"POOL0","Elapsed":1528,"Calls":0,"Wait":0.000000,"Max":0.000000,"Min":99999999.000000,"Pool Calls":0,"Pool Attempts":0,"Pool Wait":0.000000,"Pool Max":0.000000,"Pool Min":99999999.000000,"Pool Av":0.000000,"Work Had Roll Time":false,"Work Can Roll":false,"Work Had Expire":false,"Work Roll Time":0,"Work Diff":524288.00000000,"Min Diff":100000.00000000,"Max Diff":524288.00000000,"Min Diff Count":1,"Max Diff Count":2,"Times Sent":45,"Bytes Sent":7131,"Times Recv":75,"Bytes Recv":36197,"Net Bytes Sent":7131,"Net Bytes Recv":36197}],"id":1}#`

c7ph3r10, 2025-07-22:
> I would try `echo '{"command": "ascset", "param": "0,softoff,1:4"}' | nc 10.0.0.201 4028`

Still this response:

`{"STATUS":[{"STATUS":"E","When":1753220807,"Code":15,"Msg":"Missing device id parameter","Description":"cgminer 4.11.1"}],"id":1}# `

b-rowan, 2025-07-22:
You could try constructing it with pyasic if you want...

```
import asyncio

from pyasic import CGMinerRPCAPI

IP = ""

async def main():
    rpc = CGMinerRPCAPI(IP)
    print(await rpc.ascset(0, "softoff", "1:4"))

if __name__ == "__main__":
    asyncio.run(main())
```

c7ph3r10, 2025-07-22:
> You could try constructing it with pyasic if you want...

I am affraid there are some missig capabilities on my side to do this. Would really have to deep dive into pyasic to do this. Right now I just have an ssh comand prompt on my home assistance instance to shoot the commands.

c7ph3r10, 2025-07-22:
> Interestingly enough they released sources for the nano3s, this is (eventually) how those commands get handled - https://github.com/Canaan-Creative/Avalon_Nano3s/blob/04c84d312a4e38c1efa437b086bbcb297749e77a/cg_miner/cgminer/driver-avalon.c#L2907-L2933
> 
> This also contains a list of all the commands!

I asked canaan support wheather there is something like that for the avalon q´s already. Got the response that they will give it to the development team. I am affraid this will take ages.

b-rowan, 2025-07-22:
Maybe you want to try the `help` command with ascset?

`echo '{"command": "ascset", "param": "help"}' | nc 10.0.0.201 4028` or maybe `echo '{"command": "ascset", "param": "0,help"}' | nc 10.0.0.201 4028`

b-rowan, 2025-07-22:
> [@b-rowan](https://github.com/b-rowan) Did not understand what´s your issue right now? Is it just the JSON format? This seems to be solved for every "read" command but not for softoff, softon and workmode, right?

Yes kindof, but the issue is if I want to implement these functions (which are used in hass-miner) I will need to understand exactly how to format them...

b-rowan, 2025-07-22:
TL;DR: `parameter` not `param` is the key.

c7ph3r10, 2025-07-22:
> Your other commands should now work if you use `parameter` instead of `param`!

they do:

`echo '{"command": "ascset", "parameter": "0,softoff,1:4"}' | nc 10.0.0.201 4028`

response:
`{"STATUS":[{"STATUS":"I","When":1753222577,"Code":118,"Msg":"ASC 0 set info: success softoff:4","Description":"cgminer 4.11.1"}],"id":1}#`

just as mentioned: timestamp issue is still there

b-rowan, 2025-07-22:
Cool.  This gives me a starting point, I'll try to get something rolling on this tomorrow then.  Sure glad they posted source code, this will make things much easier at least to understand.

c7ph3r10, 2025-07-22:
> Cool. This gives me a starting point, I'll try to get something rolling on this tomorrow then. Sure glad they posted source code, this will make things much easier at least to understand.

Just ask any time, in case I can help with something! For sure I am also interested to have a pro version based on pyasics.
I am based in germany. Middle of the night here. Will also have to go to sleep now ;-)

b-rowan, 2025-07-22:
> For sure I am also interested to have a pro version based on pyasics.

Not sure what you mean by this, I have very little interest in making money personally off of this, I just want to help people deal with the lack of good miner management tooling.

b-rowan, 2025-07-23:
Ok, I think I have something, but it seems like they replaced the data that was in the old `stats` command.  Looking at sources, it seems like they use `minerstats` now, could you try that command?

`echo '{"command": "minerstats"}' | nc 10.0.0.201 4028`

c7ph3r10, 2025-07-23:
Command:
`echo '{"command": "minerstats"}' | nc 10.0.0.201 4028`

Response:
`{"STATUS":[{"STATUS":"E","When":1753289753,"Code":14,"Msg":"Invalid command"}],"id":1}#`

Command:
`echo '{"command": "estats"}' | nc 10.0.0.201 4028`

Response:
`{"STATUS":[{"STATUS":"S","When":1753289842,"Code":70,"Msg":"CGMiner stats","Description":"cgminer 4.11.1"}],"STATS":[{"STATS":0,"ID":"AVALON0","Elapsed":29497,"Calls":0,"Wait":0.000000,"Max":0.000000,"Min":99999999.000000,"MM ID0:Summary":"'STATS':{Ver[Q-25052801_14a19a2] LVer[25052801_14a19a2] BVer[25052801_14a19a2] HashMcu0Ver[Q_hb_v1.1] FanMcuVer[Q_fb_v1.2] CPU[K230] FW[Release] DNA[0201000018f85eaf] STATE[2] MEMFREE[67964] NETFAIL[0 0 0 0 0 0 0 0] SSID[] RSSI[0] NetDevType[0] SYSTEMSTATU[Work: In Idle, Hash Board: 1] Elapsed[29531] BOOTBY[0x01.00000000] LW[15518033] MH[2] DHW[0] HW[2] DH[2.685%] ITemp[33] HBITemp[47] HBOTemp[47] TMax[0] TAvg[0] TarT[65] Fan1[0] Fan2[0] Fan3[0] Fan4[0] FanR[0%] SoftOffTime[1753289105] SoftOnTime[0] Filter[17390] FanErr[0] SoloAllowed[1] PS[0 1220 6 0 0 2239 146] PCOMM_E[0] GHSspd[0.00] DHspd[0.000%] GHSmm[54501.36] GHSavg[66910.04] WU[934722.53] Freq[280.13] MGHS[66910.04] TA[160] Core[A3197S] BIN[36] PING[18] SoftOFF[4] ECHU[0] ECMM[0] PLL0[9564 5751 4944 4061] SF0[258 276 297 318] CRC[0] COMCRC[0] ATA0[800-65-2264-258-20] LcdOnoff[1] Activation[0] WORKMODE[0] WORKLEVEL[0] MPO[800] CALIALL[7] ADJ[1] Nonce Mask[25]}","HBinfo":"'HB0':{PVT_T0[ 57  59  58  61  60  58  60  61  63  60  60  61  61  61  59  61  66  60  61  61  60  58  59  61  63  58  59  60  61  62  61  63  61  62  60  64  61  60  60  59  62  65  65  62  63  65  66  66  67  65  68  67  69  69  69  68  68  68  69  69  69  69  69  67  67  69  68  68  69  69  69  69  68  68  66  65  65  65  64  63  60  64  65  63  64  65  69  66  66  67  70  70  69  69  71  68  70  69  70  69  70  71  70  72  71  71  70  71  71  70  68  70  67  68  67  68  67  64  66  65  65  66  63  64  60  60  59  62  63  61  60  61  65  62  63  65  63  63  63  64  63  63  63  65  63  62  66  64  62  62  64  62  61  62  62  61  62  61  58  62] PVT_V0[281 278 275 275 281 281 278 277 271 270 277 276 275 274 279 279 276 277 275 276 280 280 273 273 277 278 278 278 274 272 280 279 279 279 278 278 281 280 278 279 277 275 273 275 275 272 274 273 273 274 274 275 273 273 273 274 265 265 270 269 271 271 272 272 275 274 273 273 267 270 269 273 273 274 274 275 273 274 278 278 276 274 274 275 276 276 271 274 273 273 271 270 264 263 270 272 266 267 270 272 269 269 267 264 268 267 268 269 271 271 271 270 272 269 274 274 273 275 270 273 271 272 273 274 282 280 281 281 276 275 280 282 273 274 268 269 272 272 277 276 275 276 276 278 275 276 272 271 276 276 274 275 274 278 276 278 271 271 280 280] MW0[647 756 745 684 695 658 659 722 752 723 742 711 721 728 683 714 681 719 711 747 727 680 720 716 749 659 739 715 712 745 665 700 668 671 659 728 708 623 750 723 729 702 694 671 699 703 701 768 732 653 681 644 708 688 706 650 699 719 662 733 688 690 668 659 720 643 679 682 698 733 680 667 759 657 726 756 752 676 755 713 646 709 747 659 709 716 708 677 686 697 703 700 689 687 695 677 701 628 661 686 663 687 674 749 725 714 735 727 705 668 617 744 699 674 729 686 681 631 699 672 701 684 725 706 697 769 683 685 719 725 685 641 805 697 691 697 701 725 707 650 720 662 705 685 725 719 706 691 737 678 795 721 723 697 746 659 707 707 723 751]}"}],"id":1}#`

b-rowan, 2025-07-23:
Perfect!

c7ph3r10, 2025-07-23:
> Ok, 0.74.0 is up, hopefully it works...

Sorry for another stupid question. I am pretty new to github. Does this mean that hass-miner automatically uses version 0.74.0 or does hass-miner also have to bring out a new version?

Best case for me would be that I can install hass-miner as documented and it will recognize the avalon q´s.

ZZR600E3, 2025-07-24:
Thanks for the work everyone put in and in such a quick turnaround. I can see sensors now but I do not see any switches to control the miner. 

Is that still something yet to come with further development or is something not right on my end?

c7ph3r10, 2025-07-24:
> I wonder if I missed the shutdown flags... whoops. I'll fix it 😂

Thank you very much. I also tested it. Entities available for the new home assistant device look like this:

[image]

So, also no option to change the workmode.

c7ph3r10, 2025-07-25:
> Strange how yours has different entities active to mine, and the hash rate is showing correctly too. Can see firmware is reported as the same so not sure how both Q's have different behaviours

Yes, the "Hashrate" Entity matches to what I see as Average Hashrate in the webfrontend of the Q and "Ideal Hashrate" entity matches to the current hashrate of the frontend. Fan Speed match to what I see when I shoot the commands on CLI of my Home Assistant as well, except that there are 4 Fan values and not just 2. Nor sure what "Power Limit" has to match to. in Eco mode it shows 800 in Standard 1300 and in super 1700. The other values are still unknown.

If someone is interested: I am still working on my "workaround". But, this will be customized for the avalon q. Not tested against anyhing else and it will be home assistant specific. I put together some home assistant packages which shoot different API commands via command_line and afterwards pull out the data and figures out of the JSON which I am interested in.

In this way I will get all relevant data as ha entities which i need.

As soon as I am finished I can share the commands and the template entities yaml.

c7ph3r10, 2025-07-25:
@ZZR600E3 
I am wondering that you have three hashboard and four fan entities. What is our the firmware mentioned on the web frontend? Mine is this:

[image]

ZZR600E3, 2025-07-25:
Yep, exactly the same. Installed the version Brett published to get the Avalon Q update up and running.

DoubleDoc, 2025-07-26:
Hello, and thank you everyone helping move this along. 

Yes - In HA I had to delete HASS-MINER all the way back to the custom repository and reload to get past the ZZR600E3 reported screen (no data except a hash field) (https://github.com/UpstreamData/pyasic/issues/350#issuecomment-3116760510)

Now-The sensor data displayed is accurate, just some missing. And there are no controls.

[image]

c7ph3r10, 2025-07-27:
> So- this might not be the best repository to comment in - sound more like the HA side. But the flow was here,

@DoubleDoc 
Your are right. May be I started the mess up mixing HA and pyasic things up.

@All
As a conclusion I would like to summarize my basic research results regarding the API of the Avalon Q. May be this helps to develop pyasic for even more comprehensive support for the Canaan Avalon Q Home miner.

The device uses a cgminer implementation version 4.11.1 as of now. The API is version 3.7 and is described here:
[https://github.com/davecgh/cgminer/blob/3.7/API-README]

The API has some standard commands but also some which can be customized by the manufacturer. To the best of my knowledge the "ascset" command is the main command to customise. So did canaan. With the "help" command at least it is possible to find out which parameters are allowed for "ascset". To get to know how to handle those commands I think the only way is to sniffer network traffic while controling the device via smartphone app or web-interface. As the API is simple plain text this was easy. In this way I was able to find out the commands for controling the workmode and also the softon and softoff command. There might be some further commands for e.g. switching of LED display etc, but so far this was not too interesting for me.

**Let me know if there is something which might be interesting and I overlooked.**

At the end I tested all API 3.7 commands which I think are interesting to get data out of the device or control it. I attached a summary text file of all commands I tested with their respective JSON response. The <DEVICE IP-ADDRESS> has to be replaced by the actual ip-address.

[Avalon Q cgminer Commands (JSON).txt](https://github.com/user-attachments/files/21453911/Avalon.Q.cgminer.Commands.JSON.txt)

Following I list all data which I pulled out of the JSON responses which (a) are interesting to me or (b) I was able to reverse check against what the devices shows within the app or web-interface.

command: "COIN"
- "Network Difficulty" contains the current difficulty of the blockchain. May be interesting to show in a dashboard

command: "ASCSET" - with parameter "help"
- Shows a list of all valid commands passed via parameter to ascset. So far I only use workmode, softon and softoff. Would be interested how the commands reboot, filter-clean, time and lcd work. May be someone knows already as it is standard with other miners?

command:  "SUMMARY"
- Did not use any of the values included within the JSON response but obviously it includes some hashrate an difficulty information.

command: "STATS"
- used litestats instead but obviously the sesponse is pretty much comparable to stats

command: "LITESTATS"
- LVer equals the shown firmware version on the devices web-interface
- SYSTEMSTATU shows "In Work" when any of the workmodes Eco, Standard, Super is active and "In Idle" when the device is in standby
- ITemp: May be "inner" Temperature? Does anybody know?
- HBITemp: May be "Hash Board Inlet Temperature"? Does anybody know?
- HBOTemp: May be "Hash Board Outlet Temperature"? Does anybody know?
- TMax: Maximum Temperature since switch on(?) But which temperature? Hasboard? anywhere in the device?
- TAvg: Average temperature
- Fan1 - 4: rpm per fan. My assumption is that fan 1-2 are inlet fans and fan 3-4 outlet fans. But it´s just a guess.
- FanR: relative overall fan speed in %. So far it does not seem like fans are controlled with differend speeds as far as I compared the values over time.
- Filter: Shows a simple number which must have something to do with the duration until the filter shall be cleaned. Just have no clue what´s the pattern here. Currently I have a number of 19145 for miners which are running approx 2 month now. So it can´t be seconds. Looks more like a counter for "hours". Still have to figure out.
- FanErr: Indicates wheather a fan is faulty
- SoftOnTime: I guess this is the unix timestamp used by the app for the timer. So it might show the time when the device shall be turned on.
- SoftOffTime: I guess this is the unix timestamp used by the app for the timer. So it might show the time when the device shall be turned off.
- PS: Found out, that the last number in the list is the current load in Watt shown on the web-interface.
- GHSspd: Current gigahash speed
- DHspd: no clue about this but this one is in percent. any idea?
- PING: current latency to the connected pool
-  LCdOnoff: indicates wheather the lcd screen of the device is turned on or off. Would be interested in switching this one via ascset. any idea?
- WORKMODE: shows the active workmode. 0 = Eco, 1 = Standard, 2 = Super
There are some more datafileds but I don´t know what they are about. May be someone who had already hands on to some more miners can complete the list.

command: "POOLS"
- shows the current pool configuration. pretty much self explaining.

command: "DEVS" & "DEVDETAILS"
- shows some device status values. Did not use them so far.

command: "CHECK"
- don´t know what the JSON response should tell me. Contains "Exists: N" and "Access: N". Looks like boolean values.

command: "VERSION"
- shows data about API, CGMiner and Firmware version, etc.

command: "CONFIG"
- general information about ASIC Count, PGA Count, Pool Count, OS, etc.

command: "ascset" with Parameter "WORKMODE"
- used by the app to switch between workmodes (see attached text file for details)

command: "ascset" with Parameter "SOFTON"
- used by the app to exit standby mode. needs a unix timestamp within its parameters. (see attached text file for details)

command: "ascset" with Parameter "SOFTOFF"
- used by the app to switch to standby mode. needs a unix timestamp within its parameters. (see attached text file for details)

**EVERYTHING MORE HA RELATED I WILL POST IN THE HASS-MINER STREAM STARTING FROM HERE**
https://github.com/Schnitzel/hass-miner/issues/545#issuecomment-3117463363