UpstreamData/pyasic #370: Partially supported miner found: Antminer L9_i

Combinator78, 2025-08-16:
UserWarning: Partially supported miner found: Antminer L9_i, type: MinerTypes.ANTMINER, please open an issue with miner data and this model on GitHub (https://github.com/UpstreamData/pyasic/issues).
  warnings.warn(
ip='172.16.21.8' device_info=DeviceInfo(make=<MinerMake.ANTMINER: 'AntMiner'>, model=None, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=<class 'pyasic.device.algorithm.base.GenericAlgo'>) mac=None api_ver='3.1' fw_ver='Wed AML 21 19:20:02 CST 2024' hostname=None sticker_hashrate=None expected_hashrate=225257.0 H/s expected_hashboards=None expected_chips=0 expected_fans=None env_temp=None wattage=None voltage=None fans=[] fan_psu=None hashboards=[] config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='ltc.viabtc.io:3333', user='bloodyclaus.001', password='x'), Pool(url='ltc.viabtc.io:25', user='bloodyclaus.001', password='x'), Pool(url='ltc.viabtc.io:443', user='bloodyclaus.001', password='x')], quota=1, name=None)]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')) fault_light=None errors=[] is_mining=True uptime=270133 pools=[PoolMetrics(url=None, accepted=66410, rejected=220, get_failures=None, remote_failures=None, active=True, alive=True, index=0, user='bloodyclaus.001', pool_rejected_percent=0.33018159987993395, pool_stale_percent=0), PoolMetrics(url=None, accepted=0, rejected=0, get_failures=None, remote_failures=None, active=False, alive=False, index=1, user='bloodyclaus.001', pool_rejected_percent=0, pool_stale_percent=0), PoolMetrics(url=None, accepted=0, rejected=0, get_failures=None, remote_failures=None, active=False, alive=False, index=2, user='bloodyclaus.001', pool_rejected_percent=0, pool_stale_percent=0)] hashrate=None wattage_limit=None total_chips=None nominal=None percent_expected_chips=None percent_expected_hashrate=None percent_expected_wattage=None temperature_avg=None efficiency=None efficiency_fract=None datetime='2025-08-16T15:49:15.808759+03:00' timestamp=1755348555 make='AntMiner' model=None firmware='Stock' algo='Generic (Unknown)'
None

[image]

b-rowan, 2025-08-16:
Do you know if this is an actual new type, or just an L9 being assigned "L9_i" by the firmware?

b-rowan, 2025-08-16:
Fixed in 0.76.1

closed 2025-08-16 by b-rowan